bluerange-io/bluerange-mesh #182: Timer tick thread locks when main thread is locked

rench9, 2021-08-03:
Hey, we are performing some encoding of packets and found that until the timer tick is locked until encoding completes.
Is there any mechanism to mange threads or priorities  with fruity mesh.

mariusheil, 2021-08-03:
Hi,

take a look here for some info on which interrupts are used: https://github.com/mwaylabs/fruitymesh/blob/master/src/hal/nrf/FruityHalNrf.cpp#L112

FruityMesh runs directly on the SoftDevice and not using a RTOS so there is no direct Thread concept but only interrupt service routines. For your case, I would recommend that you use the main context handler https://github.com/mwaylabs/fruitymesh/blob/master/src/base/GlobalState.cpp#L107 which allows you to run tasks at the lowest priority possible. This will get interrupted by most of the FruityMesh logic whenever necessary so you have to make sure that you do not access any variables that you also use in other parts such as e.g. the TimerEventHandler in a thread unsafe way.

Most of the FruityMesh logic (timerEventHandler, bleEvents, .....) run in the same interrupt priority to make the processing more straight forward and less error prone.

Marius

siretty, 2021-08-31:
Hello, I'll close this as there has not been any more feedback. Feel free to reopen it.

closed 2021-08-31 by siretty