bluerange-io/bluerange-mesh #184: Is it possible to connect to non mesh (non fruityMesh) peripheral?

PriyeshNema, 2021-08-31:
Hi team,

I'm looking to connect to a 3rd party BLE peripheral (a device) from FruityMesh node. So, in my case, the FruityMesh node should act as BLE Central for this 3rd party peripheral (always). Post the connection, I also want to secure the connection using pairing and bonding. Once done, I need to read/write the characteristics of peripheral based on the data transaction on mesh network. I mean, I need to read characteristics and send the read data over mesh and on response, I need to write the characteristics based on data received from mesh.

Can someone help me if the use case I'm looking can be achieved using FruityMesh. If yes, can you provide pointers on any tutorial which can guide me in achieving this.

Thanks!

mariusheil, 2021-08-31:
Hi,

this is possible in general but will require some work. Take a look at the implementation of e.g. the MeshAccessConnection. When FruityMesh connects to a device, it first needs to know how to communicate with this device, which is why at first, a ResolverConnection will be instantiated. This ResolverConnection will then try to determine the type of communication. This is mostly done by waiting for a WRITE from the other device on some characteristic. If the other device writes on some well known characteristic, the ResolverConnection gets replaced by the correct Connection Object, e.g. a MeshAccessConnection.

Now, it looks like you want to connect to the device yourself, so you could just instantiate a new Connection with your custom type, e.g. MyCustomConnection which inherits from BaseConnection. Then, try to connect to the device using ConnectAsMaster from the ConnectionManager. You must also create your connection so that the ConnectionManager manages the instance.

Then, you should be able to get all the necesary connection events in your class and you can implement the communication.

We do not have a tutorial for this and you will need to go through the implementation and our documentation that is available on GitHub. You should be able to implement this without too much effort I would guess.

Marius

PriyeshNema, 2021-08-31:
Thanks Marius for the initial help. Will start playing around and may need your help again if required.

closed 2021-08-31 by PriyeshNema