bluerange-io/bluerange-mesh #187: Missing file when building cherrysim

Mohamed-GN, 2021-09-16:
when building cherrysim with cmake --build. it states that ErrorLog.h is missing 
i want also to know whether you can provide a standalone exe of cherrysim so i can run on my windows10

kdeneve, 2021-09-19:
I have also the problem of the missing ErrorLog.h header file when trying to build the project in VS2017 ..
Anyone can help us please? Where can we find this header file?
Greetz

mariusheil, 2021-09-19:
Hi,

thanks for reporting. These files were indeed missing from the release version. Looks like there was a mistake while uploading the latest release. I have now added the missing files and changed the tag to point to the latest commit. Please pull it again.

As for the pre-built simulator: This is not easy for us, as our buildbox is only automatically building all the unix and native arm binaries. We do not have a windows buildbox setup at the moment so we would always need to compile it manually. But compiling CherrySim should be fairly easy now, that all the files are available. Just download the latest Visual Studio Community Edition (which is free) and Cmake and it should build out of the box without any additional dependencies if you follow this guide: https://www.bluerange.io/docs/fruitymesh/BuildingWithCMake.html#BuildingSimulator

Marius

Mohamed-GN, 2021-09-19:
You can use the previous tag it worked for me.
I have some question i changed the number of in connection to 1 and out connection also to 1 so the topology will be a line but in cherrysim sometimes some nodes cant join the mesh also it seems that nodes dont make the best decision which is naturally the shortest distance how to fix this issue

mariusheil, 2021-09-19:
Hi, yes, the previous release was uploaded correctly, but you should be able to use the latest tag now.

As for your other question, it would be better to open a new issue, but just a short answer here: As for the number of connections, I have to say that we never actually tried out what happens if you set both to 1 as it would not work out to be a mesh after all. The nodes generally need 1 inConnection to be able to actually build a mesh and usually another temporary inConnection to be able to tell connection partners to cancel other connections in case they are stuck. The algorithm might not work correctly with the settings you tried and I would need to check why it behaves that way.

The algorithm is decentral and will therefore not always pick the best connections. To achieve this, a central aproach would need to be used. We have some internal tickets to implement an additional algorithm that enhances the connections over time but have not yet seen enough real-world installations where the current algorithm would not be sufficient. In general, the connections are more optimal the more time the nodes have to scan for surrounding nodes, but we have chosen the values so that they provide a  good trade-off between clusteirng time and optimized connections.
From the measurements that we have done, it does not really matter if you have a connection that has an RSSI of -70 compared to -80, the throughput you will get is comparable. It would certainly offer possibilites to optimize battery power, but that would require different TX power for each connection. All in all, there is a lot that can be done but we have not yet seen any necessity from the experience of the installations.

Marius

kdeneve, 2021-09-20:
Thanks Marius .. All working fine now 

Greetz
Kurt

Mohamed-GN, 2021-09-20:
thanks Marius i will open a new issue and i will present my problem in detail

closed 2021-09-28 by siretty