bluerange-io/bluerange-mesh #190: Support for NRF52833

DrYerzinia, 2021-09-25:
I forked and added support for the NRF52833: https://github.com/DrYerzinia/fruitymesh/tree/nrf52833

I added SDK16 as it is the first SDK to support he NRF52833 as I understand it.  I found it a bit odd that the SDK CMake file sets the Target Chip: target_compile_definitions_multi("${SDK_15_TARGETS}" "NRF52840")

Unclear to me how you are meant to use the same SDK for multiple chips.

Wondering if you want to merge in my changes for the NRF52833.  Let me know and I'll make a pull request.

siretty, 2021-09-28:
Hellow @DrYerzinia,

the SDK's CMake file sets that definition, as we only use the SDK15 for nRF52840 and the SDK14 only for nRF52832 MCUs.

Regarding the nRF52833 support and integration of the SDK16: We'll have to talk it through internally if we want to support the nRF52833 MCUs - this could take some time for which I apologize.

Kind regards
Daniel

siretty, 2021-09-29:
Hi @DrYerzinia, we would love a pull request! Thank you :smile:

siretty, 2021-10-01:
Thank you very much for the PR! We'll handle further discussion in the PR. Feel free to re-open the issue.

closed 2021-10-01 by siretty