bluerange-io/bluerange-mesh #195: Unable to save/reload bonded data

PriyeshNema, 2021-11-08:
Hi,

I have a peripheral device which only sends data if it is paired. I have enabled pairing+bonding in FruityMesh device and I'm able to pair to the peripheral.
As I reboot FruityMesh device, ideally the bond data should get reloaded, but inturn, it again starts to ask the pairing key with peripheral device again.

This did not happen if I just use nRF SDK (same version as FM) with my code. It's happeing only when FruityMesh code is integrated

Can someone help me resolve this

mariusheil, 2021-11-17:
Hi,
in order to store the bond data, you probably needed to integrate a library which stores and loads this information, correct? For this library, you also need to choose a proper place on where to store it. I imagine this could be because the FruityDeploy flash library also listens for Flash Events. There are no proper handles in the SoftDevice that would allow two libraries like this to work along as they do not know which flash event is related to which action.
We do not currently support this and I think you would need to store the bond using our library or remove the flash storage from our code. Then you should probably get bonding to work but you'll need to use the nordic libraries to store and retrieve data.

Marius

mariusheil, 2021-12-09:
Hi, I am closing this for now due to inactivity, feel free to re-open it.

closed 2021-12-09 by mariusheil