bluerange-io/bluerange-mesh #196: Reimplement FruityMesh nodes

Rewbert, 2021-12-09:
Where can I find a detailed specification of the FruityMesh protocol, so that I would be able to implement my own node as part of a larger case study on mesh networks? The premise is that I have a board with BLE capabilities, and nothing more. I read some text on the website but I am not sure if it is enough?

Thanks in advance,
Robert

closed 2021-12-09 by Rewbert

mariusheil, 2021-12-09:
Hi, not sure if you found it already, there's some text here and there is a master thesis covering the topic. But re-implementing the algorithm would be a lot of effort and is very test intensive. We have evolved the algorithm and implementation over the last 5 years, just to give you an idea. But as it is open source and there is a simulator, you will see all the implementation details in our repository:
https://www.bluerange.io/docs/fruitymesh/The-Algorithm-in-Detail.html
https://www.bluerange.io/docs/fruitymesh/CherrySim.html

Marius

Rewbert, 2021-12-09:
Hi! I did find some more text that I hadn't seen before, which seemed to contain a lot of details. Thanks for the links!

Rewbert, 2021-12-09:
@mariusheil Do you know who authored the master thesis? I can't seem to find that one.

mariusheil, 2021-12-09:
Hi,

I don't think it is publicly available, I have attached it.

[Masterthesis_Marius_Heil_2015-05-05.pdf](https://github.com/mwaylabs/fruitymesh/files/7685456/Masterthesis_Marius_Heil_2015-05-05.pdf)

mariusheil, 2021-12-09:
Hi,

I have attached it:
[Masterthesis_Marius_Heil_2015-05-20.pdf](https://github.com/mwaylabs/fruitymesh/files/7685483/Masterthesis_Marius_Heil_2015-05-20.pdf)

Marius

Rewbert, 2021-12-09:
THanks Marius!