bluerange-io/bluerange-mesh #197: LoRaWan module

bobmorane83, 2022-01-04:
Hi,

First of all, thanks for the hard work done so far !

I've compiled the Fruitymesh stack on different platform and on the RAKWireless RAK4630 and seems to work ok.
I was wondering if a Fruitymesh LoRaWan module exist ? 
I've seen the work done by RAKWireless on the RAK4630 but unfortunatly for my project, based on Arduino environment.
I'am ready to port the Semtech stack (some project exist but quite old) but if BlueRange have done some work, I would be pleased to exchange on this subject.

Thanks,
BR,

mariusheil, 2022-06-15:
Sorry, looks like this issue went under the radar. We have done some work with LoRaWan some time ago but have not followed up on this. So no, I do not know of any work in this direction, sorry.

I am closing this, but feel free to re-open it.

Marius

closed 2022-06-15 by mariusheil

bobmorane83, 2022-06-15:
No problem. 
I done it myself ;)
Best regards,