bluerange-io/bluerange-mesh #198: Is it possible to use UART without hardware flow control?

LubomirBogdanov, 2022-05-27:
Hello,
I'm evaluating fruitymesh on a custom board and so far I'm having trouble with the Terminal.cpp and UART - whenever I type a command sometimes it gets executed, sometimes it doesn't. I think this is CTS/RTS related. I have a custom featureset where I have disabled those two signals like that:

`void SetBoard_myboard(BoardConfiguration* c)
{
   ....
        c->uartCTSPin =  -1;
        c->uartRTSPin =  -1;
  ....
}`

but I'm not sure - will this turn the hardware flow control off?

Regards,
L. B.

mariusheil, 2022-05-27:
Hello, yes, this will turn flow control off. This is a general problem with the nRf Softdevice when working with UART as it will generate interrupts. You can mitigate the issue if you choose a lower baud rate or if you work with a protocol that is more fault tolerant, e.g. use CRC checksums and repeat the commands in case of error. Using easydma over UART will also increase the reception probability lot. I think we have it open sourced as well, but cannot check at the moment. Have a look if you can find a define to switch on UART with easydma. I will have a look on Monday.

Marius

mariusheil, 2022-05-30:
Hi,

I had the chance to take a look. Check FruityHalNrf.cpp around line 3200 for FH_NRF_ENABLE_EASYDMA_TERMINAL which you can use with TerminalMode::JSON

This is being used by us in one of our products and works without issues (the ones mentioned in the comment around this line are some edge cases that you can find when stress-testing the implementation).

Marius

LubomirBogdanov, 2022-05-30:
Hi,
unfortunately I had this feature already enabled and it didn't work with it. I tried to turn it off, it made no difference. 
In the file "FruityHalNrf.cpp" I edited FruityHal::EnableUart( ) to lower the baudrate but it didn't work either...

mariusheil, 2022-05-30:
Mhh, hard to say without seeing the issue. You are definitely not getting an echo-back of your input? Once you enter the command and enter the line feed, it should either execute the command or give you an error that it could not understand the command. Depending on the configuration it will also print out the entered command that was received. If it does not execute on the first try, try to enter the line feed sequence once more and see if your command was received. You can also use SEGGER RTT to see what is being received and what is not. You could enter some SEGGER_RTT_printf statements to check if this is the issue or you are having a different issue.

If you work with a virtual com port such as e.g. a SEGGER debugger providing the virtual com port, make sure it is working as intended as the SEGGER virtual COM port is rather basic and has a lot of trouble with flow control and sometimes requires switching the device on and off.

Using EasyDMA should work quite well in general.

Marius

mariusheil, 2022-06-15:
I am closing this because of inactivity, feel free to re-open it.

closed 2022-06-15 by mariusheil