bluerange-io/bluerange-mesh #199: How to change advertising interval

LubomirBogdanov, 2022-06-01:
Hello everyone,
I'm trying to evaluate the current consumption of a fruitymesh node. So far I measure 1.63 mA average current (using a battery simulator unit, and not a multimeter, so it should be very accurate). I have turned the UART and TERMINAL off from the featureset.

Is there a way to change the advertising interval?

In a Nordic project I would change the APP_ADV_INTERVAL macro, but in fruitymesh I can only see this macro in nrf_ble_dfu.c which is (probably) related to the bootloader and not the application.

Regards,
L. B.

mariusheil, 2022-06-15:
Hi,

the used advertisement, connection, etc,.... intervals are specified in the Config.h and Config.cpp and you can change them, but be aware that too much change might result in bad connection properties as we have tried to figure out good values already.

The most power consumption will come from the mesh connections. Changing the connection interval will have a significant effect.

If you are interested in low power nodes, you should make sure that the discovery will be turned off after all nodes have been connected. Now, the difficult part is to know what "all nodes" really means. If you have a gateway, or know the number of nodes that must be connected, you can use the "set_enrolled_nodes" feature (https://www.bluerange.io/docs-internal/fruitymesh/Node.html#_setting_number_of_enrolled_nodes).

Marius

mariusheil, 2022-09-15:
Hi, I will close this for inactivity, feel free to re-open it.

closed 2022-09-15 by mariusheil