bluerange-io/bluerange-mesh #200: Long term connection intervals (CONN_PARAM_UPDATE)

arzippi, 2022-07-19:
Hi,

I'm currently trying to minimize power consumption for mesh nodes. 
These are the steps I already tried:
- `set_enrolled_nodes` from a gateway -> `DiscoveryState::IDLE` consumes around 400 uA
- Increasing `meshAdvertisingIntervalLow` (Config.h) from 200 to 500 ms

I stumbled upon the `CONN_PARAM_UPDATE` define, but I'm not 100% sure what the impact is. I doubled the `meshMaxLongTermConnectionInterval` (Config.cpp) to 30ms and activated the define. This results in a power consumption around 200 uA. 

Since I wasn't able to find any documentation on this, can you give some more insight into what the intention of this define is? From what I understand for my setup, it means that the connection parameters will change, after the mesh connection was successfully accomplished. Because I don't need maximum data rate or minimal delay, I can happily live with slower connection parameters if it saves almost half the power consumption.

mariusheil, 2022-08-09:
Hello,

sorry, your Issue went under the radar. Is it still an issue? I have found that we have some documentation on "CONN_PARAM_UPDATE", but only internal. As you already found out, the idea is to have a long term connection interval to reduce power consumption in the mesh. so, aber a while once the mesh is established, we use lower intervals to save power. You can look for "IS_ACTIVE(CONN_PARAM_UPDATE)" for some comments. This implementation is not used a lot by us so make sure to report issues if you find them. There are however some tests in CherrySim from which you can see how it is being used, take a look at "TestConnectionParameterUpdateRequested" in TestNode.cpp.

Otherwhise, your changes sound good. What we haven't used so far is slave latency on the connections, which you could also implement as this will further reduce the consumption. However all the nodes that have to keep connections as a central will not benefit from this so it is hard to predict on which nodes you will save power.

Marius

mariusheil, 2022-09-15:
Hi, I will close this for inactivity, feel free to re-open it if necessary

closed 2022-09-15 by mariusheil

arzippi, 2022-09-15:
Hi Marius,

thanks for the additional documentation.
Working with the `CONN_PARAM_UPDATE` works as expected and so far I haven't found any issues. I will look further into the `Conf::meshConnectionLongTermAgeDs` mentioned in the attached documentation, however, for my application it is absolutly fine if the connection parameters are changed immediately. 

Thanks again for your support.