bluerange-io/bluerange-mesh #201: How to make it support the Acconeer XM122

hlwdy, 2022-08-16:
Hello, I am a Chinese high school student who loves computers. Nice to see you on GitHub! We are trying to do a project on the IoT and had some difficulties. We are very interested in the project FruityMesh. As we are new to embedded development, we do not know much about this area.
    We have tried to run FruityMesh on nrf52840-dk and it's great! We now want to change the code to make it support Acconeer XM122 (The XM122 is built around the Nordic Semiconductor nRF52840 Bluetooth® 5 SoC and it features an integrated antenna for Bluetooth connectivity).
    I wonder if this idea is feasible? How should it be realized? Could you please give me some suggestions on how to modify the code to make it support the new board?

mariusheil, 2022-08-16:
Hello,

getting FruityMesh running on a different PCB such as the Xm122 is easy. First thing you need to do is to get the datasheet of your board. It seems like there is an nRF52840 on board and another chip, a radar chip is connected through SPI. You will need to replace the firmware on the nRF52840 using the SWD pins (SWDIO, SWCLK), see page 8 of the datasheet (https://developer.acconeer.com/download/xm122-datasheet-pdf/).
Next, you should create your own board definition file. For the quick and dirty way, you can modify board_18.cpp as this is the board that is loaded by default once you compile FruityMesh for the nRF52840 chipset. You can see that it is loaded in Boardconfig.cpp where SetBoard_18 is called.
The proper way would be to flash the correct boardId in the UICR as mentioned here: https://www.bluerange.io/docs/fruitymesh/Developers.html#UICR

Now, first you should have some safe defaults, so use dcDcEnabled false and set the clock source to CLOCK_SOURCE_RC and the accuracy to CLOCK_ACCURACY_500_PPM. You can then change these values according to the datasheet or you can just start with these and develop and change them later.

For debugging, use Segger RTT as your UART will not be working in the beginning until all the pins and settings are correct. Getting the LED pins correctly assigned (however your board might not have any). All the pins that you need to use are in the column "nRF52840* pin" in the datasheet.

Finally, you probably need to get the radar chip working, therefore you'll need to integrate the driver provided by Acconeer, which means using SPI communication to communicate with the radar chip. We have done this with a few different sensors, but have not yet used a radar chipset, so you need to figure out a few things yourself, but if the driver is well written, it should be easy to integrate.

Marius

hlwdy, 2022-08-16:
I have got the Acconeer XM122 and the Acconeer XB122. The XM122 is connected to the XB122 and XB122 is connected to my PC by an USB cable. I have noticed none of these boards contains a Segger J-Link debugger chip. Maybe I can't use Segger RTT for debugging?
As for the last part, I have got the radar working. So what I'm just curious about is how to make the FruityMesh working on my board.
I have already created the board definition file and I think the pins are correct. But what should I do to replace the firmware on the nRF52840 using the SWD pins (SWDIO, SWCLK)? (the first step that you mentioned)

hlwdy, 2022-08-16:
I'm not sure if I did exactly what you said. I have already changed the board definition and compiled. I also programmed the board using DFU. Unfortunately, it seems that it doesn't work correctly and I didn't get any output on the serial port with UART.
Could you please tell me what should I do?

mariusheil, 2022-08-16:
Hi,

the boards themselves don't have a segger debugger chipset, but you can wire the SWD pins of your nRF52840 DK so that you can program the external board and Segger RTT will work over the same wires.
You could get FruityMesh on the board using DFU but this will not allow you to do proper development so you should flash FruityMesh using SWD.
For UART, all the pins and the baud rate / parity / flow control need to be correct, so it will be easier for you to test with Segger RTT first as it has less pitfalls. Try to get the LED working as well for testing if the pin assignment works as expected.

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_nrf52840_dk%2FUG%2Fdk%2Fhw_debug_out.html&cp=4_0_4_7_9

It is hard to provide more help as this is usually a hands-on process.

Marius

hlwdy, 2022-08-16:
Ok. Thank you very much! This answers some of my questions. I think I probably know how to do it and I will have a try.

closed 2022-09-15 by mariusheil