bluerange-io/bluerange-mesh #202: How to activate the logging for the sdk

Tobi15, 2022-10-25:
I am currently trying to use an SD card to store data on from the fruitymesh layer. 

I have added the missing libraries in sdk 15 to use the fatfs functionality. Now my program works but reboots when some methods in the sdk are called. The problem is that I don't have a log in the terminal and I have no idea where the error came from... 

To facilitate development, I need to see the sdk log.  I tried to set the NRF_LOG_ENABLED 1 flag in sdk_config.h but it doesn't work. 

Have you a basic setting to enable the logging from the nrf5 sdk v15? 

Best regards

Dominik-Helfenstein, 2022-11-02:
Hi,

is it possible for you to connect an RTT Viewer terminal? If so, you can enable fruitymesh's logging inside the featureset, e.g. in featureset `github_dev_nrf52.h` there are defines for logging and to enable Segger RTT, which enables fruitymesh's terminal logging. If you have this enabled like in the featureset `github_dev_nrf52.h`, you can connect Segger RTT Viewer and view some output.
You can also debug over UART if you don't have access to the RTT Viewer.

With sdk-wide logging I'm not deep enough into the topic right now, maybe you can use it somehow with RTT too.

If I have misunderstood your question feel free to clarify.

Hope I could help a little bit.

Tobi15, 2022-11-16:
Hi, 

The Fruitymesh terminal logging is already enabled. I can see all logt() that are written in the Fruitymesh layer. 
I can also see the logs from the HAL like shown just below : 
```
[FruityHalNrf.cpp@4618 DEBUG]: Initializing disk 0 (SDC)…1
[FruityHalNrf.cpp@4622 DEBUG]: Initializing disk 0 (SDC)…
[FruityHalNrf.cpp@4625 DEBUG]: try to init the disk!! DISK_STATE : 1
[FruityHalNrf.cpp@4625 DEBUG]: try to init the disk!! DISK_STATE : 1
[FruityHalNrf.cpp@4625 DEBUG]: try to init the disk!! DISK_STATE : 1
[FruityHalNrf.cpp@4630 DEBUG]: Disk initialization failed.
```

As you can see, I only got the FruityMesh log. It would be interesting to see the sdk logs. For example, when in the HAL, I call the method
```
disk_state = disk_initialize(0);
```
I would like to see logs that are inside the method disk_initialize(0) which is a method that comes from the sdk.

Best regard,

dippa-1, 2024-03-14:
Hi,

Sorry for not answering.
Have you managed to activate SDK logging?
Otherwise I can look into it.

closed 2024-03-19 by dippa-1