bluerange-io/bluerange-mesh #203: Ruuvi Tag B8

phaethon, 2022-11-02:
Hello!

Is Ruuvi Tag rev. B8 supported by this firmware? I am reading that some sensors have changed from B7.1. Will I be able to read sensor data with this firmware?

Thanks in advance!

Dominik-Helfenstein, 2022-11-02:
Hi,
as of now, not without writing some sensor interfaces.
They added the TMP117 and DPS310, which both don't have sensor interfaces in fruitymesh, sorry

phaethon, 2022-11-02:
If those 2 drivers are implemented, then everything else should work?

Dominik-Helfenstein, 2022-11-03:
1) You need those two drivers.

2) Then you can include, configure, control them inside a module (`src/modules`).
There already is a `RuuviWheatherModule.h` and `RuuviWheatherModule.cpp` which you can copy adjust everything that's relevant to you (like the 2 new sensors).

3) This module then has to be used by a featureset (currently there is `prod_ruuvi_weather_nrf52` inside `config/featuresets`).
You can copy the existing featureset and change https://github.com/mwaylabs/fruitymesh/blob/eee8ce7094a75e52bedb9807ec4a6a80ff478094/config/featuresets/prod_ruuvi_weather_nrf52.cpp#L114
to initialize your newly created module.

Then it hopefully works

I hope I could help

closed 2024-03-14 by dippa-1