bluerange-io/bluerange-mesh #205: CherrySim unable to build

mabner1996, 2023-02-21:
Hello, I just downloaded the newest version of FruityMesh and would like to try the new CherrySim simulator
However, I found that I cannot build the CherrySim properly
[image]
As seen in the screenshot, there is a syntax error in parameters constant of the GetReceptionRssiNoNoise function.

Dominik-Helfenstein, 2023-02-22:
Hey,

I could not reproduce your issue.
I did the following steps:
1. Installed Visual Studio 2022 Community Edition with C++ build tools
1. Cloned this repository
1. Created a folder called `cherrysim` next to `fruitymesh`
1. Opened the `cherrysim` folder in PowerShell
1. Ran the command `cmake ../fruitymesh -A Win32 -DBUILD_TYPE=SIMULATOR`
1. Opened the newly generated Visual Studio solution with `./CherrySim.sln`
1. Compiled and ran CherrySim with and without debug mode

From the screenshot I can tell that you are using an older Visual Studio version.
- Please try it with a newer version of Visual Studio
- Have you changed the version of C++ to an older one that doesn't support this kind of syntax?
- Have you changed any other settings inside Visual Studio?

Let me know if you have further questions

mabner1996, 2023-02-23:
I was using visual studio 2017, the issue has been resolved with VS2022, thanks!

closed 2023-02-23 by mabner1996