bluerange-io/bluerange-mesh #207: Will it be updated ? When?

yuyangworld, 2024-03-06:
The project codes have not been updated for a long time.So is it dead?

dippa-1, 2024-03-06:
Hi,

It's definitely not dead!
We are about to release a new feature very soon

yuyangworld, 2024-03-07:
@dippa-1 Thank you very much for the response.Does it have a roadmap?

dippa-1, 2024-03-07:
Not an official one but the next feature is the AutoSense module. It can be configured to poll a number of registers/component ids from any module. For example, this could be configured to poll a temperature sensor every 5 seconds and automatically publish a `component_sense` message with the sensor data when the data has changed.

As of future modules, there are two more that are almost done, but I can't comment on these yet.

Out of curiosity, would you share what you are working on / planning to work on? It's always interesting to see projects with FruityMesh.

yuyangworld, 2024-03-08:
@dippa-1 I am going to port it to other BLE SOC,such as ESP32.It's better to be based on the latest version.I expect to enhance the max number of BLE connections by FruityMesh.

dippa-1, 2024-03-08:
That's great to hear. Feel free to keep us updated and to submit a PR. We plan to release the latest sources in the coming days. However, it is unlikely that it includes any changes that effect porting to a different SOC.

dippa-1, 2024-03-14:
The lastest release is now published

closed 2024-03-14 by dippa-1