bluerange-io/bluerange-mesh #211: How to connect Android devices to bluerange mesh？

BigBooks, 2025-02-20:
This is really a great project, especially the networking algorithm, I think it is very powerful. Thanks for the open source!

1. I would like to ask, I now want to connect Android devices to the mesh and participate in the networking algorithm, not just as a controller.
What I mean is, is it possible for mobile phones to use this networking algorithm to communicate between mobile phones?

2. Is it difficult to transplant this algorithm on Android? Or do you have any suggestions for the development of mobile phone network communication algorithms?

I'd be very grateful for any useful help. Thanks！

BigBooks, 2025-02-20:
Or is there an event flow chart? I roughly understand that this is the implementation of a set of event triggering mechanisms, but I am not very clear about the logic of event triggering because I do not have the hardware. But I very much hope to do something with this set of networking algorithms.

mariusheil, 2025-02-20:
Hi,

it is possible to connect mobile phones to the mesh nodes and communicate with the netwok using the MeshAccessConnections. We haven't tried to port the algorithm to android itself. There are some hurdles in the way, such as the encryption (paring/bonding) and the simultanious usage of both the peripheral and central mode of BLE. Also, the BLE stacks on mobile platforms are not always stable as we've encountered in some projects. Sometimes they tend to lose connectivity that in the worst case require an OS restart, e.g. that happened for use when trying to handle a large number of connection requests. It is however not impossible as the BLE stacks have gotten better at handling the mentioned things.

On the difficulty, it's hard to say. With AI, you might be able to port a lot of the code. The only thing that needs actual porting would be the FruityHal, the other code is quite generic and should work on most platforms (if you use a native SDK that supports c/C++).

As for the flow chart, try to set up CherrySim. This allows you to test out the mesh without any actual hardware :-)
https://bluerange.io/docs/bluerange-mesh-community/CherrySim.html

There's quite a bit of documentaton to start and some flowcharts, e.g. here:
https://bluerange.io/docs/bluerange-mesh-community/The-Algorithm-in-Detail.html

Be aware that this documentation is quite old and some things might have changed. So using CherrySim would be a good way to start.

Marius

BigBooks, 2025-02-21:
Thank you very much for your reply, I will learn CherrySim

BigBooks, 2025-02-21:
Hi, I have a question, does the only logic for new nodes to access the network depend on CalculateClusterScoreAsMaster and CalculateClusterScoreAsSlave？
What I mean is, does the networking algorithm consider the optimal network topology to ensure that communication between nodes only requires the smallest possible number of hops?

BigBooks, 2025-03-04:
Hi, I am trying to transplant recently, but I am not very clear about how to actually connect, network, and transmit data to bluerange. Can you provide me with a log including: networking, data transfer and other operations?
Thanks.